nsiatras/sourcerabbit-gcode-sender #1: soft reset button creates new window

alpharesearch, 2015-04-08:
Sometimes pressing the soft reset button creates new window, but both windows are usable?

nsiatras, 2015-04-09:
Hello,

Are you sure is this happening?
There is no method in the code that re-opens the window.

nsiatras, 2015-04-09:
If you are running the GCode Sender from code:

Inside frmMain there is a method OpenControlForm.
This method closed frmMain and opens frmControl and removes the fConnectionEstablishedEventListener.

Just check if this is happening

alpharesearch, 2015-04-09:
Here is a private video of the issue: http://youtu.be/nSEtekoV6t4
I do run it from Netbeans 8.0.2, but it only happens when if fails to connect first... I guess something is not cleaned up properly... I only added this line:
jComboBoxPort.addItem("/tmp/ttyV1");
so I don't have to type my com port in all the time...

nsiatras, 2015-04-09:
Hello,

Thank you for the video. It was very helpful!
I just fixed the bug and pushed it to GitHub.

Thank you !

alpharesearch, 2015-04-09:
works good now

closed 2015-04-09 by alpharesearch

nsiatras, 2015-04-09:
You are welcome!