nsiatras/sourcerabbit-gcode-sender #2: Can't change baud rate

rhfinch, 2017-07-25:
The opening window appears to have a drop-down list to change baud rate, but the only option is 15200. Is there any way I can change to 9600?
Thanks

nsiatras, 2017-07-26:
The Baud rate is hardcoded inside the frmMain code line 88 **jComboBoxBaud.addItem("115200");**
May I ask why you want to change it ?

rhfinch, 2017-07-26:
I have compiled grbl to run at 9600 baud so I can optionally connect to it from my android phone via bluetooth using an HC-06 baord. I can then use an android app like CNC commander for testing my newly built CNC router without have to connect a PC.

itxitx, 2017-10-31:
me too... can't set the baud rate... all other gcode senders i tried have an option to change the baud rate. 
so, basically it's useless because i can't even connect

nsiatras, 2017-11-13:
Hello,

SourceRabbit GCode Sender is build in order to work properly with our machines.
For that reason we have set the baud rate to 115200 

Thank you

closed 2018-01-02 by nsiatras