nsiatras/sourcerabbit-gcode-sender #3: None of the dropdowns have any options

ipetepete, 2017-09-01:
I don't have any options, and can't start the app.
I've tried building from source, running as root and nothing...

I also created the directory `/var/lock` as described on the Readme.

OS: Mac Sierra
Java: 
java version "1.8.0_25"
Java(TM) SE Runtime Environment (build 1.8.0_25-b17)

nsiatras, 2017-09-02:
Hello,

Do you have any arduino device connected to your computer?

ipetepete, 2017-09-02:
Yes. The device is present (Arduino Uno *clone*)

port: /dev/cu.wchusbserial1420

I can send commands using grblController

nsiatras, 2017-09-02:
On your first message you said you dont have any options and the app does
not start.
 When you run the app does a "connect to device" window comes up ?

ipetepete, 2017-09-02:
Yes, I get the initial window with the three drop downs. They are all empty, and I can't get past that point.

nsiatras, 2017-09-02:
Can you set a breakpoint in frmMain.InitUI and see if you get an exception there ?

ipetepete, 2017-09-05:
Sorry for the late reply. 

Looks like there isn't any exception, but `String[] serialPorts = SerialPortList.getPortNames();` Is empty. So I changed that line to: `String[] serialPorts = SerialPortList.getPortNames("/dev/", Pattern.compile("cu.*"));` Which found my serial devices. If I have time I might write a condition which will get the port names list in this manner when run on a Mac.

nsiatras, 2017-09-05:
I am not very sure that the problem is the getPortNames condition. I tested the software on Mac and I know several people that use in on Mac without any problems. Maybe you will find more info on JSSC github page as JSSC is responsible to identify the serial connections on the computer.

May I ask if you are using original MacOS or Hacintosh ?

ipetepete, 2017-09-05:
I'm on an Macbook Air 2012 running Sierra 10.12.6

smartynov, 2017-12-03:
I experience the same issue. All dropdowns are empty. Same Sierra 10.12.6 on MacBook Pro 2017.

nsiatras, 2018-02-23:
The bug is now fixed

closed 2018-02-23 by nsiatras