nsiatras/sourcerabbit-gcode-sender #4: Can't connect

dastultz, 2017-11-20:
Hello, I have a Mac 10.13.1, GRBL 1.1f with Java 1.8.0_25. I get the following when I try to connect. There are no other options from the ones selected. I'm confident it's the correct port and it's functional. Is there any logging done? What could be wrong? My GRBL installation works fine with UGCS.

Thanks.

[image]

nsiatras, 2017-11-20:
Hello,

If port is correct then close any other software, like UGS, that are using the port

dastultz, 2017-11-20:
The port is not in use. I even rebooted. It looks like both your project and UGS use JSSC for serial, and UGS is working fine (the 7/23 nightly build anyway).

nsiatras, 2017-11-20:
Have you tried this ?

_Note for MAC users: You may need to create a "/var/lock" directory with write permission. To do this open the Terminal application and run the following two commands: 
sudo mkdir /var/lock 
sudo chmod 777 /var/lock_

dastultz, 2017-11-20:
Yes, UGS has the same suggestion and as mentioned UGS is connecting fine.

```
OTWKLT0076:MacOS daryl$ ls -l /var
lrwxr-xr-x@ 1 root  wheel  11 Nov 12 21:48 /var -> private/var
OTWKLT0076:MacOS daryl$ ls -l /private/var | grep lock
drwxrwxrwx   17 root          _uucp          544 Nov 12 21:59 lock
```

nsiatras, 2017-11-20:
Do you have an official arduino with FTDI chip or a clone with the CH340 ?

dastultz, 2017-11-20:
I have a Sanguino with a (version 1) USB-BUB from Modern Device. The chip has an FTDI logo on it.

nsiatras, 2018-01-02:
Hello,

Did you manage to connect ?

dastultz, 2018-01-02:
Hi, no I didn't. I've committed to using Universal GCode Sender so I've not tried again.

closed 2018-01-02 by dastultz

nsiatras, 2018-01-02:
Really sorry to hear that and really sorry.
Can you please tell me what baud rate you use with the UGS ?

dastultz, 2018-01-02:
115200