nsiatras/sourcerabbit-gcode-sender #5: Problem with current and previous version hole centre

tubaian, 2018-04-13:
The hole centre finder will not work for me. When I initiate it, my mill jerks once and doesn't move. The script seems to think it is still running and the script window usually closes when I hit the 'stop' button in it but several times the whole program has locked up so that I've had to kill the whole thing. I'm running SourceRabbit 1.2.0 on Linux Mint and the latest GRBL. I also downloaded and tried version 1.1.9 but had the same results.

nsiatras, 2018-04-16:
Sounds like a machine ground issue.
Make sure the touch probe's pin is connected to the GND.

If this does not work for you then insulate the touch probe pin

tubaian, 2018-04-16:
Thanks Nikos. I was using an insulated probe and also insulated the work from the machine bed and connected one lead of the probe input to each. I am now thinking that it might be an issue with GRBL as sending
 a G38.2 X20 F60 from the command line only gets a twitch from the machine as well and no movement. Probing the Z height works perfectly its just the hole centre I can’t get to work and I have a piece of work on the machine that I need to set up accurately and finish the second side..

Ian

nsiatras, 2018-04-16:
Hello Ian,

What version of GRBL you have in your controller ?

tubaian, 2018-04-17:
Hi Nikos, Sorry I took a while to reply - I’m running GRBL 1.1f

Ian

haaxema, 2018-05-27:
I think I found the "problem". I discribed it in https://github.com/nsiatras/sourcerabbit-gcode-sender/issues/8

tubaian, 2018-06-02:
Hi, I have no limits enabled on my grill but the hole probe still doesn’t work.....

nsiatras, 2018-06-02:
Just to make sure we are on the same page. The problem you all have is very common.
If the touch probe is not insulated then you will never be able to find center of a hole.

By insulated I mean that if you take your multimeter, set it to continuity and check **the continuity between stepper motors, spindle and touch probe, the result must be negative.**

haaxema, 2018-06-02:
My probe works without any problem when I use it for Z probing. Isolation is not the issue!

tubaian, 2018-06-06:
> Can you share a photo of your "insulated" touch probe ?

Hi, here is a photo. As you can see, the probe is a piece of carbide rod with 3 layers of heat shrink tubing over the end that goes in the collet. The aluminium plate I’m working on is insulated from the machine bed by a piece of wood and the hold downs are onto cardboard. The two connections to my probe input are the crocodile clip and the piece of printed circuit board to the left. Even with no electrical connections to the work, the X axis only makes a slight judder when I tell it to find the hole centre. I have also yesterday tried BCNC which has a similar routine and that did make an effort to work but I think I need to learn a bit more about its setup as it didn’t perform correctly, however, it did move the X axis either way and then the Y axis similarly and responded to the probe being touched by the PC board - it just didn’t make it all the way to the sides of the hole….[cid:5FC1B87E-03CA-45FA-B43D-AB5C7DBF18AD]

nsiatras, 2018-06-06:
Unfortunately I cannot see the photo. If the machine just makes a  slight judder when you start the hole center finder sequence I believe it is a wiring problem. SourceRabbit GCode sender is build to be compatible with the SourceRabbit machines and hardware so I really don't know how to help you make them similar to our setup. Just make sure you connect the Carbide rod to the GND and the other end to the SCL pin of the CNCShield

tubaian, 2018-06-06:
Thanks Nikos,

My point was that, even though Z-probing works fine with my setup, the hole probing doesn’t move the X and Y axes even with nothing electrically connected to the probe input on the cnc shield. Anyway, the job is now so far behind that I am going to have to try to find another solution to my problem. Thanks again..  Ian

closed 2020-01-12 by nsiatras