nsiatras/sourcerabbit-gcode-sender #6: Gcode sender software problem

kusmanf, 2018-05-09:
Sir, I am a beginner to CNC machines. I made myself a CNC machine and need to use it as a plotter. I use Arduino Nano with V4 shield and A4988 drivers for NEMA 17 stepper motors. 
I installed Arduino software and did GbrlUpload too. Selected the correct serial COM port(3) and selected Baud as 115200. 
After connecting to the PC when I browse my Gcode file (it is a spiral sketch)created with Inkscape and hit the send button, under Machine Status/Active State indicate "Idle". Time elapsed advances. When I opened Visualize I cannot see any movement or graphic. For your information I did not connect motors  to the Arduino nano board. Is it the problem? Please reply me and advise me. Thank you. Brgds. Fernando

nsiatras, 2018-05-09:
Try to upload GRBL using the sketch files
https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino#for-windows

Generate a GCode from the following link
http://cncwebtools.com/Apps/Table_Milling_GCode_Generator/index.html

Make sure you connected motors and power in a proper way

And then send the GCode

kusmanf, 2018-05-20:
Nsiatras,
Thanks for the advice. I shall try your suggestion.
Brgds.
Fernando

closed 2019-11-28 by nsiatras