nsiatras/sourcerabbit-gcode-sender #11: Requests

3dPrintingMD, 2018-10-19:
1) Ability to Maximize the Window, currently grayed out
2) Step Sizes - Would be great if you could have the ability to select step size from a few choices.
3) Probing - When that is running I do not see any gcode displayed, I feel that would be helpful.

Thanks - Going to run it on my cnc and see how it runs.

skendrot, 2019-03-13:
Would be great to get separate step sizes for X/Y and Z

nsiatras, 2020-01-12:
@3dPrintingMD 

1. You can maximize the control window in the latest version of the software. 

2. We can certainly do that but we have to think about a useful way to implement that in the UI

3. The only reason we do not display the live position of the machine during a touch probe operation is because we do not want to use more of the Arduino's CPU in order to maintain precision

klembaris, 2020-03-29:
Hello,
would be nice to have edge finder (probing) XYZ in the future releases.

nsiatras, 2020-04-28:
@3dPrintingMD 

Your request about the step sizes is now implemented on our latest release
You can select step sizes of 0.001, 0.01, 0.1, 1, 10 and 100

justin-hubbard, 2020-05-25:
I'd like the ability to set zero points for each axis individually, so i can use different methods of setting my zero points. Currently there is one button for setting the current (x,y,z) zero points all at once, I would appreciate an additional button for each axis to set them one at a time. Thanks for taking requests!

nsiatras, 2020-05-25:
Double click on the X,Y and Z under Work Position.
It will ask you if you want to zero each axis individually

justin-hubbard, 2020-05-25:
Oh gotcha. Just checked and I see how to do it. For a future version, some buttons for that functionality would be nice, and easier to find. I never would have found those functions without asking.

closed 2021-03-29 by nsiatras