nsiatras/sourcerabbit-gcode-sender #12: Stops sending gcode in the middle of program.

MCuserme, 2018-12-28:
I have an issue that it stops sending gcode to a controller after a random time, mostly on bigger gcode files. 
I noticed when in that state I press pause and than resume again it sends one line of code.

It is very frustrating because it destroys a workpeace since it does not remember the last coordinates.

I really like this software because it is only one that runs ok on my weak laptop, so if somone can help I would be very grateful.

nsiatras, 2018-12-28:
Hello,

What software and what post-processor are you using to generate the gcode?

MCuserme, 2018-12-28:
I use Aspire and LaserWeb, both of them have the same issue. 
It does not stop on the same file always at the same spot...

nsiatras, 2018-12-28:
Ok. Your problem has not to do with the software.

I don't know your machine setup but the most common issue is an electromagnetic interference between your controller and some other device. 

Examples of devices that creates high electromagnetic forces:  
Fluorescent lamps
Vacuum cleaners and other motors
Cheap Chinese quite cut spindles
etc..

PS. What machine do you have ?

nsiatras, 2018-12-28:
Can you share a gcode file ?

MCuserme, 2018-12-28:
[laser_engrave.zip](https://github.com/nsiatras/sourcerabbit-gcode-sender/files/2714862/laser_engrave.zip)
I had to put it in the zip. since github wont let me upload .gcode.

nsiatras, 2018-12-28:
How much bytes out / second do you have when you send the GCode to your
machine?

ulab, 2019-10-13:
I had noticed something similar today. I have just started using Rabbit. 

On OSX it feels like it stops or slows down sending g-code after a while when I switch to other apps (a video for example). As soon as I switch back, it sends data again.

nsiatras, 2020-01-12:
@MCuserme Download the latest version of our Software and upgrade your GRBL to the latest version (version 1.1h).  We did a lot of optimization since the last time and I believe you will run your GCode without any problems

closed 2020-04-28 by nsiatras