nsiatras/sourcerabbit-gcode-sender #16: Set machine position

skendrot, 2019-04-15:
Need a way to set the machine position, not just the working position and it needs to persist. Software should not start at 0,0 every time.

ulab, 2019-10-13:
The 'machine position' is set when you run the home cycle and is only reset when you either home again or reset GRBL (which will require a homing cycle). 

I'm new to this myself, but so far I have not seen any other way to set the machine position.

All your workflow should be based on the working origin.

nsiatras, 2019-11-29:
We have to "hack" the GRBL firmware to be able to change the machine's positionbvalues from the software.

So unfortunately this is not possible at the moment

closed 2020-01-12 by nsiatras